dss-extensions/OpenDSSDirect.py #130: class_to_dataframe is returning dict now?

nlaws-camus, 2023-12-08:
### Versions

- [ ] Python version: Python 3.10.9
- [ ] Operating system and version: Ubuntu 22.04.3 LTS
- [ ] OpenDSSDirect.py version number: 0.8.4

### Bug
```python
dss.utils.class_to_dataframe("Load")
```
is returning a dict like
```
{'Load.load_p4ulv5_1': {'phases': '1', 'bus1': 'p4ulv5.1', 'kV': '0.12', 'kW': '2.27376117064562', 'pf': '0.962717604705656', 'model': '1' .... }} 
```
We did not change versions so maybe something changed in the underlying C-API?? 🤷

PMeira, 2023-12-09:
No, that is assembled in Python. Do you Pandas installed? It's an optional dependency -- if importing it isn't successful, dicts are return instead of dataframes.

If you do have it installed in this environment, would you mind listing the packages so we can try to reproduce it? If you don't already use a `requirements.txt` or similar, `python -m pip freeze > requirements.txt` should be enough.

nlaws-camus, 2023-12-18:
I am using [bazel](https://bazel.build/) to manage environments so maybe something changed in bazel, but adding `requirement("pandas")` to the BUILD file resolved the issue. Thank you @PMeira !

closed 2023-12-18 by nlaws-camus