dss-extensions/OpenDSSDirect.py #133: Bug in LineGeometry interface

AadilLatif, 2024-03-26:
### Versions

- Python version: 3.11
- Python architecture: x64
- Operating system and version: Windows 11
- OpenDSSDirect.py version number: 0.9.3

### Bug
LineGeometries methods are not providing the correct information see example below

Opendss model :

New LineGeometry.OH3P_FR8_N56_OH_477_AAC_OH_336_AAC_ABCN nconds=4 nphases=3 reduce=yes
~ cond=1 wire =OH_477_AAC x=-3.67 h=31.33 units=ft
~ cond=2 wire =OH_477_AAC x=0 h=31.33 units=ft
~ cond=3 wire =OH_477_AAC x=3.67 h=31.33 units=ft
~ cond=4 wire =OH_336_AAC x=0 h=26 units=ft

odd.LineGeometries.Conductors() - > [OH_477_AAC, OH_477_AAC, OH_477_AAC]
odd.LineGeometries.Nconds() -> 3

odd.Text.Command(f"? LineGeometry.{geometry_name}.nconds")
odd.Text.Result() -> 4

PMeira, 2024-03-26:
Found the issue: currently it returns `pLineGeometry.Nconds`, which is a Pascal property, includes a check for `reduce`, returning the number of phases if it's enabled:

https://github.com/dss-extensions/dss_capi/blob/0.14.4/src/General/LineGeometry.pas#L754-L760

But, since `reduce` is just a flag, not an action (it doesn't modify other input data etc.), that check doesn't make sense for the API users.

I'll try to prepare new packages for tomorrow.

The AltDSS version, which is derived from the property system, returns the expected values (4, and 4 names/objects), so that's an alternative.

PMeira, 2024-03-29:
The updated package is available on PyPI. A new test was added, but let me know if any issues remain.

closed 2024-03-29 by PMeira