dss-extensions/OpenDSSDirect.py #136: bus names are set to lower case: can this be prevented?

nlaws-camus, 2024-11-30:
### Support Question
I am programmatically defining all circuit elements based on data that can have mixed string cases, and later I need to align the bus names (and many other things in the dss model) with the data. Is there a way to stop dss from changing the capitalization of the bus names?
```python
dss(f"""
Clear
new circuit.dummy Bus=CapsBUS
solve
""")
dss.CktElement.BusNames()
```
['capsbus', 'capsbus.0.0.0']

PMeira, 2024-12-11:
Currently, no. Nearly everything in OpenDSS/AltDSS is case-independent, so keeping the original names would either require maintaining copies or applying a upper/lowercase conversion in a lot of places. That is, currently not worth the effort of considering.

We can keep this open as a feature request though. We might be able to handle it cleanly in a few months (after the full C++ transition on the engine).

nlaws-camus, 2024-12-19:
Any tips on how to deal with indexing into the `dss.Circuit.YNodeOrder`, which does not change the node names,  using `dss.CktElement.BusNames()`, which are lower case?

For example, to get the source nodes one can
```python

  source_busses: list[str] = []
  source_bus_phases: list[list[int]] = []
  dss.Circuit.SetActiveClass("VSource")

  dss.Vsources.First()

  def _update_output():
      if dss.CktElement.Enabled():
          # BusNames is like ['source', 'source.0.0.0'] for a single 3 phase
          # source, and can look like ['newbus.1', 'newbus.0.0.0'] for a
          # single 1 phase source. 
          bus_name = dss.CktElement.BusNames()[0]
          # NOTE dss.Bus.Name() does not update with the VSource
          if "." in bus_name:
              bus_name = bus_name.split(".")[0]
          source_busses.append(bus_name)
          nodes = set(dss.CktElement.NodeOrder())
          nodes.discard(0)  # drop ground connections
          source_bus_phases.append(list(nodes))

  _update_output()
  while dss.Vsources.Next() > 0:
      _update_output()

  source_nodes = [
      bus + "." + str(phase)
      for bus, phases in zip(source_busses, source_bus_phases)
      for phase in phases
  ]
```

But the `YNodeOrder` will have the capitalization that was used in the `New` commands, while  `BusNames` has lower case names.

PMeira, 2024-12-19:
> But the `YNodeOrder` will have the capitalization that was used in the `New` commands, while `BusNames` has lower case names.

Maybe we could at least add an option to return lowercase strings from `YNodeOrder`. 

Related to your sample, in the upcoming release, to avoid this common pattern...

```python
if "." in bus_name:
              bus_name = bus_name.split(".")[0]
```

...`BusNames`  now has an argument (keyword argument in ODD.py) to suppress the nodes.

I personally don't like that the bus connections are kept as strings, even in the internal code, but I haven't had the time to try alternatives yet.

nlaws-camus, 2025-11-18:
I mistook the bus names in `YNodeOrder` for matching the bus names passed to construction commands like `New.Line` because my raw bus names were all caps. Turns out `YNodeOrder` upper-cases bus names while other methods appear to lower-case names:
```python
dss.Basic.ClearAll()
dss("""
new circuit.test
new Line.lin1 bus1=BuS1 bus2=Bus2
""")
dss.Solution.Solve()
dss.Lines.First()
dss.CktElement.BusNames()
```
```sh
['bus1', 'bus2']
```
```python
dss.Circuit.YNodeOrder()
```
```sh
['SOURCEBUS.1',
 'SOURCEBUS.2',
 'SOURCEBUS.3',
 'BUS1.1',
 'BUS1.2',
 'BUS1.3',
 'BUS2.1',
 'BUS2.2',
 'BUS2.3']
```