dss-extensions/OpenDSSDirect.py #139: Exception Raised While Processing DSS Command: Show Voltages LN Nodes

HafizRiaz, 2025-06-28:
Hi there, i am new to use openDSS with python. i was trying to run a repo, but it is giving the error as follows, 

Unable to create file "G:\Mehboob\Stable_DDPG\IEEE13Nodeckt_VLN_Node.txt": The process cannot access the file because it is being used by another process.

Probable Cause:
Error in command string or circuit data.
The snapshot of Panapower PowerShell  prompt is attached for reference. I am wondering if you may help me resolve this issue 

[image]

PMeira, 2025-06-29:
> The process cannot access the file because it is being used by another process.

@HafizRiaz, since you don't say it explicitly, have you checked if the file being used by another process? If so, you need to close the other program. For example, if the file is open on Excel, Notepad, or another editor, you could get this error message. If `G:` is a cloud drive (Google Drive, for example), weird stuff also happens -- in that case, try either setting the OpenDSS datapath to a normal folder in your local storage, or just move the `Stable_DDPG` folder somewhere else and try again.

HafizRiaz, 2025-06-29:
Thanks for the prompt response.
Actually, these files are being automatically generated and opened in note pad during training or testing and error message is displayed.  Which is perhaps happening due to show command of opendsss which i couldn't trace. May be due to call of an internal opendss function.

PMeira, 2025-06-29:
If you don't want to change the Python code, you can also set the environment variable `DSS_CAPI_ALLOW_EDITOR`. On cmd: `set DSS_CAPI_ALLOW_EDITOR=0`, and on PowerShell: `$env:DSS_CAPI_ALLOW_EDITOR=0`. Setting it as a system or user env variable through the Windows GUI would also work, but remember that it only affects new processes.

I should also mention that Notepad itself doesn't seem to grab exclusive access of the file, so you should be having issues due to Notepad. Maybe an antivirus is still scanning the file when a new report is being requested. If you're still having issues after the editor is disabled, I can try to list some tentative instructions to find more info.

HafizRiaz, 2025-06-29:
Thansk a lot. The issue has been resolved now.

________________________________________________________

Engr. Hafiz Mehboob Riaz
Lecturer, Electrical Engineering Department (E.E.D)
University of Engineering & Technology (U.E.T) Taxila, Pakistan
Mobile # 0092-302-7259033
Office: 051-9047-311

PMeira, 2025-06-30:
Searched a bit, from the file names in your traceback, it seems you're trying to run the following repo, right? https://github.com/JieFeng-cse/Stable-DDPG-for-voltage-control/

From a quick look, there is no version mentioned anywhere, so maybe there's something related to that.

The paper doesn't even mention that OpenDSSDirect.py/DSS-Extensions was used (just "OpenDSS" is mentioned), let alone list the versions of anything. Common issue, but quite annoying, both for reproducibility and lack of proper credits.

That said, I gave it a try and managed to run `test.py` on both Windows and Linux. On Windows, I did this on a cmd window:

```batch
conda create -p c:/temp/ddpg python=3.8 numpy pandas scipy pytorch pandapower matplotlib-base numba
conda activate c:/temp/ddpg
pip install dssdata gym torch
git clone https://github.com/JieFeng-cse/Stable-DDPG-for-voltage-control/
cd Stable-DDPG-for-voltage-control
python test.py
```

Maybe there's something wrong with your env. If you want me to take a look, please list the output from `conda list` on your env. Here's the output I get:

```
# packages in environment at c:\temp\ddpg:
#
# Name                    Version                   Build  Channel
brotli                    1.1.0                h2466b09_3    conda-forge
brotli-bin                1.1.0                h2466b09_3    conda-forge
brotli-python             1.0.9            py38hd3f51b4_8    conda-forge
bzip2                     1.0.8                h2466b09_7    conda-forge
ca-certificates           2025.6.15            h4c7d964_0    conda-forge
certifi                   2024.8.30          pyhd8ed1ab_0    conda-forge
cffi                      1.17.0           py38h4cb3324_0    conda-forge
charset-normalizer        3.4.0              pyhd8ed1ab_0    conda-forge
cloudpickle               3.1.1                    pypi_0    pypi
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
contourpy                 1.1.1            py38hb1fd069_1    conda-forge
cycler                    0.12.1             pyhd8ed1ab_0    conda-forge
deepdiff                  8.0.1              pyhd8ed1ab_0    conda-forge
dss-python                0.15.7                   pypi_0    pypi
dss-python-backend        0.14.5                   pypi_0    pypi
dssdata                   0.1.7                    pypi_0    pypi
filelock                  3.16.1                   pypi_0    pypi
fonttools                 4.53.1           py38h4cb3324_0    conda-forge
freetype                  2.13.3               h57928b3_1    conda-forge
fsspec                    2025.3.0                 pypi_0    pypi
gym                       0.26.2                   pypi_0    pypi
gym-notices               0.0.8                    pypi_0    pypi
h2                        4.1.0              pyhd8ed1ab_0    conda-forge
hpack                     4.0.0              pyh9f0ad1d_0    conda-forge
hyperframe                6.0.1              pyhd8ed1ab_0    conda-forge
idna                      3.10               pyhd8ed1ab_0    conda-forge
importlib-metadata        8.5.0              pyha770c72_0    conda-forge
importlib-resources       6.4.5              pyhd8ed1ab_0    conda-forge
importlib_resources       6.4.5              pyhd8ed1ab_0    conda-forge
intel-openmp              2024.2.1          h57928b3_1083    conda-forge
jinja2                    3.1.6                    pypi_0    pypi
kiwisolver                1.4.5            py38hb1fd069_1    conda-forge
lcms2                     2.17                 hbcf6048_0    conda-forge
lerc                      4.0.0                h6470a55_1    conda-forge
libblas                   3.9.0           32_h641d27c_mkl    conda-forge
libbrotlicommon           1.1.0                h2466b09_3    conda-forge
libbrotlidec              1.1.0                h2466b09_3    conda-forge
libbrotlienc              1.1.0                h2466b09_3    conda-forge
libcblas                  3.9.0           32_h5e41251_mkl    conda-forge
libdeflate                1.24                 h76ddb4d_0    conda-forge
libffi                    3.4.6                h537db12_1    conda-forge
libfreetype               2.13.3               h57928b3_1    conda-forge
libfreetype6              2.13.3               h0b5ce68_1    conda-forge
libhwloc                  2.11.2          default_hc8275d1_1000    conda-forge
libiconv                  1.18                 h135ad9c_1    conda-forge
libjpeg-turbo             3.1.0                h2466b09_0    conda-forge
liblapack                 3.9.0           32_h1aa476e_mkl    conda-forge
[... 46 lines cut ...]
scipy                     1.10.1           py38h1aea9ed_3    conda-forge
setuptools                75.3.0             pyhd8ed1ab_0    conda-forge
six                       1.16.0             pyh6c4a22f_0    conda-forge
sympy                     1.13.3                   pypi_0    pypi
tbb                       2021.13.0            h62715c5_1    conda-forge
tk                        8.6.13               h2c6b04d_2    conda-forge
torch                     2.4.1                    pypi_0    pypi
tqdm                      4.67.1             pyhd8ed1ab_0    conda-forge
typing-extensions         4.13.2                   pypi_0    pypi
ucrt                      10.0.22621.0         h57928b3_1    conda-forge
unicodedata2              15.1.0           py38h91455d4_0    conda-forge
urllib3                   2.2.3              pyhd8ed1ab_0    conda-forge
[... cut ...]
xorg-libxau               1.0.11               hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xz                        5.8.1                h208afaa_2    conda-forge
xz-tools                  5.8.1                h2466b09_2    conda-forge
zipp                      3.21.0             pyhd8ed1ab_0    conda-forge
zstandard                 0.19.0           py38h91455d4_0    conda-forge
zstd                      1.5.7                hbeecb71_2    conda-forge
```

HafizRiaz, 2025-06-30:
Thanks for your kind response and consideration.
Yes I was trying to run the same repo you are talking about.
Here is the output of conda list. I will mentioned here that, I was using the same environment to runb another project which requires pandapower and some other packages as well. If you may find some hints to dig throughmy issue. It will be a great help from your side.

# packages in environment at C:\Users\sa\anaconda3\envs\myenv:
#
# Name                    Version                   Build  Channel
about-time                4.2.1                    pypi_0    pypi
absl-py                   2.0.0                    pypi_0    pypi
alive-progress            3.2.0                    pypi_0    pypi
asttokens                 2.4.1                    pypi_0    pypi
astunparse                1.6.3                    pypi_0    pypi
ca-certificates           2023.12.12           haa95532_0
cachetools                5.3.2                    pypi_0    pypi
certifi                   2023.11.17               pypi_0    pypi
cffi                      1.16.0                   pypi_0    pypi
charset-normalizer        3.3.2                    pypi_0    pypi
cloudpickle               3.0.0                    pypi_0    pypi
colorama                  0.4.6                    pypi_0    pypi
coloredlogs               15.0.1                   pypi_0    pypi
comm                      0.2.2                    pypi_0    pypi
contourpy                 1.2.0                    pypi_0    pypi
cycler                    0.12.1                   pypi_0    pypi
debugpy                   1.8.6                    pypi_0    pypi
decorator                 5.1.1                    pypi_0    pypi
deepdiff                  6.7.1                    pypi_0    pypi
dm-tree                   0.1.8                    pypi_0    pypi
dss-python                0.14.4                   pypi_0    pypi
dss-python-backend        0.13.4                   pypi_0    pypi
dssdata                   0.2.0                    pypi_0    pypi
et-xmlfile                2.0.0                    pypi_0    pypi
exceptiongroup            1.2.0                    pypi_0    pypi
executing                 2.0.1                    pypi_0    pypi
filelock                  3.13.1                   pypi_0    pypi
flatbuffers               23.5.26                  pypi_0    pypi
fonttools                 4.47.2                   pypi_0    pypi
fsspec                    2023.12.2                pypi_0    pypi
gast                      0.5.4                    pypi_0    pypi
gin-config                0.5.0                    pypi_0    pypi
google-auth               2.26.2                   pypi_0    pypi
google-auth-oauthlib      1.2.0                    pypi_0    pypi
google-pasta              0.2.0                    pypi_0    pypi
grapheme                  0.6.0                    pypi_0    pypi
grpcio                    1.60.0                   pypi_0    pypi
gym                       0.23.0                   pypi_0    pypi
gym-notices               0.0.8                    pypi_0    pypi
h5py                      3.10.0                   pypi_0    pypi
humanfriendly             10.0                     pypi_0    pypi
idna                      3.6                      pypi_0    pypi
imageio                   2.35.1                   pypi_0    pypi
importlib-metadata        7.0.1                    pypi_0    pypi
importlib-resources       6.1.1                    pypi_0    pypi
ipykernel                 6.29.5                   pypi_0    pypi
ipython                   8.18.1                   pypi_0    pypi
jedi                      0.19.1                   pypi_0    pypi
jinja2                    3.1.3                    pypi_0    pypi
jupyter-client            8.6.3                    pypi_0    pypi
jupyter-core              5.7.2                    pypi_0    pypi
keras                     2.15.0                   pypi_0    pypi
kiwisolver                1.4.5                    pypi_0    pypi
libclang                  16.0.6                   pypi_0    pypi
llvmlite                  0.39.1                   pypi_0    pypi
markdown                  3.5.2                    pypi_0    pypi
markupsafe                2.1.3                    pypi_0    pypi
matplotlib                3.8.2                    pypi_0    pypi
matplotlib-inline         0.1.6                    pypi_0    pypi
ml-dtypes                 0.2.0                    pypi_0    pypi
mpmath                    1.3.0                    pypi_0    pypi
nest-asyncio              1.6.0                    pypi_0    pypi
networkx                  3.2.1                    pypi_0    pypi
[... 54 lines cut ...]
termcolor                 2.4.0                    pypi_0    pypi
tf-agents                 0.19.0                   pypi_0    pypi
torch                     2.2.0+cu118              pypi_0    pypi
torchaudio                2.2.0+cu118              pypi_0    pypi
torchvision               0.17.0+cu118             pypi_0    pypi
tornado                   6.4.1                    pypi_0    pypi
tqdm                      4.66.1                   pypi_0    pypi
traitlets                 5.14.1                   pypi_0    pypi
typing-extensions         4.8.0                    pypi_0    pypi
tzdata                    2023.4                   pypi_0    pypi
urllib3                   2.1.0                    pypi_0    pypi
vc                        14.2                 h21ff451_1
[... cut ...]
werkzeug                  3.0.1                    pypi_0    pypi
wheel                     0.41.2           py39haa95532_0
wrapt                     1.14.1                   pypi_0    pypi
zipp                      3.17.0                   pypi_0    pypi
________________________________________________________

Engr. Hafiz Mehboob Riaz
Lecturer, Electrical Engineering Department (E.E.D)
University of Engineering & Technology (U.E.T) Taxila, Pakistan
Mobile # 0092-302-7259033
Office: 051-9047-311

PMeira, 2025-07-01:
Regarding your env, I reproduced most of it but still couldn't reproduce any of the issues you've found so far. You could try creating another env and reinstalling the packages. I noticed your packages all come from PyPI, so you could try running `pip freeze > requirements.txt`  and reinstalling the packages on the new env with `pip install -r requirements.txt`. If you still have issues even on the new env, it could be something release to your locale settings or other settings on your machine.

Specifically to ODD.py, your list includes 0.8.x, and you could update it to the latest release which does include bugfixes (an upcoming release is also pending some fixes will be out when I have some time). I don't think it affects IEEE13 results.

> Would you suggest me some relevant repos or any existing work related to this which I may use for my work. Any positive response from your side will be a great help for me please.

Specifically to DDPG, not sure, but there is **a lot** of activity in the field in general.

If you're not looking for something specific, try tweaking this search: https://www.google.com/search?q=python+gym+opendss+site:github.com&tbs=li:1 -- try some variations of gym/gymnasium/learning and so on.

Some notables repositories:

- https://github.com/NREL/PowerGridworld 
- https://github.com/siemens/powergym/
- https://github.com/tamu-engineering-research/OpenGridGym
- https://github.com/kevinrussellmoy/gym-opendss -- this one uses COM, but can be trivially ported to DSS-Python
- https://github.com/dgloves/DSS-Gymnasium -- this one doesn't have a license in the repository, but seems more up to date

The issue with older repos is that they might be using outdated versions, and upgrading may require some work. 

We plan to add some examples of how to use DSS-Extensions for better performance on larger scale systems, using multi-threading and sharing some data across solutions. For your current goal, IEEE13 is small enough that it probably doesn't help much to invest on those.

HafizRiaz, 2025-07-02:
Dear Paulo Meira,
Thank you very much for your detailed and helpful response. I truly appreciate the time you took to test and clarify things, and your insights have been incredibly useful.
I’ll follow your suggestion to create a fresh environment using pip freeze and will also upgrade opendssdirect to the latest version. It does seem like the convergence and file access issues could be system-specific, so this will help isolate the problem.
I’m also exploring the repositories you shared — especially PowerGridWorld and OpenGridGym — for future extensions of my volt-var control work.
If you happen to add more examples to DSS-Extensions for volt-var or RL-related use cases in the future, I’d love to learn from them as well.
Thanks again for your support and your contribution to the community.

________________________________________________________

Engr. Hafiz Mehboob Riaz
Lecturer, Electrical Engineering Department (E.E.D)
University of Engineering & Technology (U.E.T) Taxila, Pakistan
Mobile # 0092-302-7259033
Office: 051-9047-311

  *   https://github.com/NREL/PowerGridworld
  *   https://github.com/siemens/powergym/
  *   https://github.com/tamu-engineering-research/OpenGridGym

HafizRiaz, 2025-07-27:
Dear Paulo Meira,
I'm writing to you as I've found your SUGGESTED OpenDSS-Python repositories incredibly helpful for getting started. Thank you for making them available!
I'm currently working on plotting phase-wise voltages for the IEEE 123-bus system, but I'm encountering some issues with getting accurate results.
I've included the relevant code snippet below. If you have a moment to review it and could offer any hints or suggestions, I would greatly appreciate it.

def get_all_bus_voltages():
    """Return phase-wise voltages for all buses (skip missing phases)."""
    buses = dss.Circuit.AllBusNames()
    phaseA, phaseB, phaseC = [], [], []
    for bus in buses:
        dss.Circuit.SetActiveBus(bus)
        v = dss.Bus.PuVoltage()  # [Va_real, Va_imag, Vb_real, Vb_imag, Vc_real, Vc_imag]
        # Phase A
        phaseA.append((v[0]**2 + v[1]**2)**0.5 if len(v) >= 2 else None)
        # Phase B
        phaseB.append((v[2]**2 + v[3]**2)**0.5 if len(v) >= 4 else None)
        # Phase C
        phaseC.append((v[4]**2 + v[5]**2)**0.5 if len(v) >= 6 else None)
    return phaseA, phaseB, phaseC, buses

# =========================
# 📊 Helper: Plot voltages
# =========================
def plot_voltage_profiles(baseline, controlled, buses):
    """Plot per-phase voltage profiles for baseline vs controlled."""
    baseline_A, baseline_B, baseline_C = baseline
    controlled_A, controlled_B, controlled_C = controlled

    fig, axs = plt.subplots(3, 1, figsize=(12, 10), sharex=True)
    phases = ['A', 'B', 'C']
    colors = ['red', 'green', 'blue']
    baselines = [baseline_A, baseline_B, baseline_C]
    controlleds = [controlled_A, controlled_B, controlled_C]

    for ax, phase, color, base, ctrl in zip(axs, phases, colors, baselines, controlleds):
        x = np.arange(1, len(buses) + 1)

        # Only plot non-None values
        base_vals = [b if b is not None else np.nan for b in base]
        ctrl_vals = [c if c is not None else np.nan for c in ctrl]

        ax.plot(x, base_vals, 'D', color=color, label=f'Baseline (Q=0)', alpha=0.6)
        ax.plot(x, ctrl_vals, '^', color=color, markerfacecolor='none',
                label=f'DDPG Controlled', alpha=0.9)

        ax.axhline(1.05, color='gray', linestyle='--', linewidth=0.8)
        ax.axhline(0.95, color='gray', linestyle='--', linewidth=0.8)
        ax.set_ylabel(f'Phase {phase} Voltage (p.u.)')
        ax.legend()

    axs[-1].set_xlabel("Bus Number")
    plt.suptitle("Voltage Profiles (Baseline vs DDPG-Controlled)")
    plt.tight_layout()
    plt.savefig("voltage_profiles.png")
    print("✅ Voltage profile plot saved: voltage_profiles.png")

    # Save voltages to CSV for further analysis
    with open("voltage_profiles.csv", "w", newline="") as f:
        writer = csv.writer(f)
        writer.writerow(["Bus", "Base_A", "Ctrl_A", "Base_B", "Ctrl_B", "Base_C", "Ctrl_C"])
        for i, bus in enumerate(buses):
            writer.writerow([
                bus,
                baseline_A[i] if baseline_A[i] is not None else "",
                controlled_A[i] if controlled_A[i] is not None else "",
                baseline_B[i] if baseline_B[i] is not None else "",
                controlled_B[i] if controlled_B[i] is not None else "",
                baseline_C[i] if baseline_C[i] is not None else "",
                controlled_C[i] if controlled_C[i] is not None else ""
            ])
    print("✅ Voltage data saved to voltage_profiles.csv")
________________________________________________________

Engr. Hafiz Mehboob Riaz
Lecturer, Electrical Engineering Department (E.E.D)
University of Engineering & Technology (U.E.T) Taxila, Pakistan
Mobile # 0092-302-7259033
Office: 051-9047-311

  *   https://github.com/NREL/PowerGridworld
  *   https://github.com/siemens/powergym/
  *   https://github.com/tamu-engineering-research/OpenGridGym